dalefugier/DOSLib #1: Compiling for AutoCAD 2021?

truevis, 2020-01-31:
What will we have to do to make an ARX that works with AutoCAD 2021? We can get both the AutoCAD 2021 beta and ObjectARX SDK from https://feedback.autodesk.com/
[image]

dalefugier, 2020-01-31:
The next version of AutoCAD is a binary breaking release. Since I don't support pre-releases, you have two options: 1.) clone the repository, and modify and build project against the next ObjectARX SDK yourself, or 2.) wait for the next version of AutoCAD to released, at which time I will to 1.).

truevis, 2020-02-12:
>... Since I don't support pre-releases, you have two options: 1.) clone the repository, and modify and build project against the next ObjectARX SDK yourself, or 2.) wait for the next version of AutoCAD to released, at which time I will to 1.).

Good news! Are there instructions anywhere on how to compile the DOSLib src into an ARX with Visual Studio 2017 and AutoCAD 2020?

dalefugier, 2020-03-27:
I've updated the source to support AutoCAD 2021. You can find a pre-built binary in under Releases.

closed 2020-03-27 by dalefugier