dalefugier/DOSLib #4: DOSLibLoader

truevis, 2020-05-14:
The _DOSLibLoader_ function in the CHM file uses 'acad' as a variable. Checking code gives "redefinition of built-in symbol: ACAD". Suggest to change to 'acadv'.

(defun DOSLibLoader  (/ acadv proc fname); Determine the version of AutoCAD
  (setq acadv (substr (getvar "ACADVER") 1 2))
                                        ; Determine the system's processor architecture
  (setq proc (= "AMD64" (getenv "PROCESSOR_ARCHITECTURE")))
                                        ; Build a file name string
  (if proc
    (setq fname (strcat "DOSLib" acadv "x64.arx"))
    (setq fname (strcat "DOSLib" acadv ".arx")))
  (if (findfile fname)
    (arxload fname nil)
    (prompt (strcat "\n" fname " not found.")))
  (princ))

Pierre-de-la-Verre, 2020-11-04:
Is it solved with @ #9 ?

dalefugier, 2020-11-04:
Hi @truevis, @Pierre-de-la-Verre:

The CHM documentation is slowing being moved over to a Wiki. 

https://github.com/dalefugier/DOSLib/wiki/Introduction#loading-doslib-automatically

-- Dale

closed 2020-11-04 by dalefugier