dalefugier/DOSLib #10: dos_difference: no lists longer than 3 entries

Pierre-de-la-Verre, 2020-11-04:
In lists with more than 3 values it counts only the first three elements and brings "nil" or a wrong result without warning

Taken from Help-file:

Command: (dos_difference '(10 3 5 8))
2 

Taken from TheSwamp:
http://www.theswamp.org/index.php?topic=56360.msg602213#msg602213

: (dos_difference '(1 2 3 4) '(10 3 5 8))
nil
: (dos_difference '(1 2 3) '(10 3 5))
(-9 -1 -2)
: (dos_difference '(10 3 5))
2
: (dos_difference '(10 3 5 1000))
2

dalefugier, 2020-11-04:
Let me know if this helps:

https://wiki.mcneel.com/doslib/dos_strformatdate

-- Dale

closed 2020-11-04 by dalefugier

Pierre-de-la-Verre, 2020-11-06:
Yes, it seem to be the same problem:
```
Befehl: (dos_difference '(10 3) '(2 7))
(8 -4)

Befehl: (dos_difference '(10 3 20 30) '(2 7 5 6))
nil

Befehl: (dos_difference '(10.0 3 20 30) '(2 7 5 6))
(8 -4 15 24)

Befehl: (dos_difference '(10 3 20 30 8 9 10 11) '(2 7 5 6))
nil

Befehl: (dos_difference '(10 3 20 30 8 9 10 11) '(2 7 5 6 2 3 4 5))
(8 -4 15 24 6 6 6 6)

Befehl: (dos_difference '(10 3 20) '(2 7))
nil

Befehl: (dos_difference '(10 3 20 30 8 9 10) '(2 7 5 6 2 3 4 ))
(8 -4 15 24 6 6 6)

Befehl: (dos_difference '(10 3 20 30 8 9) '(2 7 5 6 2 3))
(8 -4 15 24 6 6)

Befehl: (dos_difference '(10 3 20 30 8 ) '(2 7 5 6 2 ))
(8 -4 15 24 6)

Befehl: (dos_difference '(10 3 20 30 ) '(2 7 5 6  ))
nil

Befehl: (dos_difference '(10.0 3 20 30 ) '(2 7 5 6  ))
(8 -4 15 24)

Befehl: (dos_difference '(10.5 3 20 30 ) '(2 7 5 6  ))
(8.5 -4 15 24)

Befehl: (dos_difference '(10 3 20 30 ) '(2 7 5.0 6  ))
nil

Befehl: (dos_difference '(10 3.0 20 30 ) '(2 7 5.0 6  ))
nil

Befehl: (dos_difference '(10.0 3 20 30 ) '(2 7 5.0 6  ))
(8 -4 15 24)

Befehl: (dos_difference '(10.0 3 20 30 ) '(2 7 5 6  ))
(8 -4 15 24)

Befehl: (dos_difference '(10 3 20 30 ) '(2.0 7 5 6  ))
nil

Befehl: (dos_difference '(10.0 3 20 30 ) '(2 7 5 6  ))
(8 -4 15 24)
```