dalefugier/DOSLib #12: dos_isinsidecurve - wrong results in some cases?

Pierre-de-la-Verre, 2020-11-10:
Hi

I found that in some cases - depending on the coordinates of the point and on the form of the curve - the result of this functions gives NIL (out), although it should be TRUE (inside).

Find attached the 
- link for a screencast: https://autode.sk/3loloxC 
- the relating drawing
- a simple lisp used for simple usage of the function.

I'm not sure on which side of the keyboard the problem resides ..

Have a good time!
[dos_isinsidecurve.zip](https://github.com/dalefugier/DOSLib/files/5518714/dos_isinsidecurve.zip)

dalefugier, 2020-11-10:
Hi @Pierre-de-la-Verre,

If you move your curves close to the world origin, does the function work as expected?

-- D

Pierre-de-la-Verre, 2020-11-11:
Hi @dalefugier 

I made more tests:
a) With Autocad with high coordinates, like above: there are missing some lines in different curve
b) With Autocad with low coordinates (1000, 8000): there is no result in one special curve
c) With Bricscad: same as b)

a) and b) are documented in a (long) screencast; start to approx. 12 min shows chaptera), the following chapter b)
https://autode.sk/35hmwNQ

c) is documented here:
https://autode.sk/2UeYSv9

I attach the DWG with low coordinates

[dos_isinsidecurev_low_coordinates.zip](https://github.com/dalefugier/DOSLib/files/5523143/dos_isinsidecurev_low_coordinates.zip)

Roy-043, 2020-11-11:
Hi,

Another issue:
The algorithm uses, what I will call, the 'count intersections with a single ray' method. This method fails when the point lies on an extended edge of the boundary.

See here:
http://ww3.cad.de/foren/ubb/Forum145/HTML/000602.shtml#000008
And see here for a potential solution:
https://www.theswamp.org/index.php?topic=48399.msg546187#msg546187

In the attached dwg (zipped) the origin is reported as no lying inside the polyline.

[single_ray_test.zip](https://github.com/dalefugier/DOSLib/files/5523780/single_ray_test.zip)