dalefugier/DOSLib #13: Suggestion: Reuse last position of dialogues

Pierre-de-la-Verre, 2020-11-16:
If this is not the correct place for suggestions please let me know.
------------------------------
It seems that all the dialogues (esp. the message boxes) appears always centred, not using the last position where they were closed before. Is there a change to reuse the last position? Or will it be implemented sometimes?

dalefugier, 2020-11-16:
Hi @Pierre-de-la-Verre,

You are welcome to drop wishes here.

DOSLib dialog boxes that inherit from `CAcUiDialog` will remember their last positions. I don't recall if all the message boxes (there are more than one) use a custom dialog or just hack into the Windows message box.

-- Dale

Pierre-de-la-Verre, 2020-11-18:
Hi @dalefugier 

thanks for reply,but now I don't know - will there be changed something or can the user do some special tricks?

dalefugier, 2020-11-18:
Hi @Pierre-de-la-Verre,

DOSLib is open-source. So if you want to fix this  yourself, you can clone the project, make a branch, change the code, and submit a pull request. Otherwise, I'll try to remember to have a look at this the next time I'm in the code.

Thanks,

-- Dale

Pierre-de-la-Verre, 2020-11-18:
Thanks Dale
my coding competence would not really improve this project ;-)
For me it is OK this way, it is no high-level priority ..

closed 2021-03-12 by dalefugier