dalefugier/DOSLib #18: About dos_popupmenu

KuoTze, 2021-09-03:
recent
Instruction dos_popupmenu
After a period of use
Will not be able to display pop-menu
In other words, he can't show it
Causes the command to be unable to follow the prompt of dos_popupmenu
Let AUTOLISP proceed to the next stage of work
I don't know how to troubleshoot this problem

I have this problem when I execute AutoCAD2021&2022 on windows10 & windows11
This has not happened in the past few years

dalefugier, 2021-09-03:
Can you provide me a simple script that allows me to repeat what you are seeing?

KuoTze, 2021-09-04:
(defun C:XX (/ :pt: :ss: :os-dim:)    

                 (setvar "cmdecho" 0)
	       (setvar "DYNMODE" 0)
                (setq :idgn_fun:
		      '("< iDesign Xline »²§U½u¥\¯àªí >"
			""
			"Ultra Xline > Horizontal"
			"Ultra Xline > Vertical"
			""
			"Ultra Xline > Horizontal + Vertical"
			""
			"Ultra Line by Layer > Horizontal"
			"Ultra Line by Layer > Vertical"
			""
			"Ultra Line by Layer > Horizontal + Vertical"
			""
			"Ultra Xline:Layer > On"
			"Ultra Xline:Layer > Off"
			""
			"Ultra Xline:Layer > Lock"
			"Ultra Xline:Layer > unLock"
			""
			"Purge Ultra Xline Object"
			""
			"< iOffset >"
		       )
	       )

	       (setq :vv: (dos_popupmenu :idgn_fun:))  ;;; After a period of time, POP-MENU cannot be displayed ~~

               (cond ((= :vv: 0) (princ))
		     ((= :vv: 1) (:idgn_x-xline:))
		     ((= :vv: 2) (:idgn_y-xline:))
		     ((= :vv: 3) (:idgn_xy-xline:))
		     ((= :vv: 4)
		      (while (setq :pt: (getpoint (:idgnmsg: "Point for X-axle.line" " : ")))
		        (if(and :pt: (or(acet-sys-control-down)(acet-sys-shift-down)) )
		          (cond
		            ((and :pt: (acet-sys-shift-down)  ) (setvar "OSMODE" 0)(:idgn_y-axline:)(:idgn_x-axline:)(setvar "OSMODE" :os-dim:))
		            ((and :pt: (acet-sys-control-down)) (setvar "OSMODE" 0) (:idgn_y-axline:)(setvar "OSMODE" :os-dim:))
			   ;((and :pt: (acet-sys-control-down)(acet-sys-shift-down))(:idgn_xy-axline:)(setvar "OSMODE" :os-dim:))
		          )
		          (progn(setvar "OSMODE" 0)(:idgn_x-axline:)(setvar "OSMODE" :os-dim:))
		        )
		      )
		     )
		     ((= :vv: 5)
		      (while (setq :pt: (getpoint (:idgnmsg: "Point for Y-axle.line" " : ")))
		        (if(and :pt: (or(acet-sys-control-down)(acet-sys-shift-down)) )
		          (cond
		            ((and :pt: (acet-sys-shift-down)  ) (setvar "OSMODE" 0) (:idgn_y-axline:) (:idgn_x-axline:)(setvar "OSMODE" :os-dim:))
		            ((and :pt: (acet-sys-control-down)) (setvar "OSMODE" 0) (:idgn_x-axline:) (setvar "OSMODE" :os-dim:))
		          )
		          (progn(setvar "OSMODE" 0)(:idgn_y-axline:)(setvar "OSMODE" :os-dim:))
		        )
		      )
		     )
		     ((= :vv: 6) (:idgn_xy-axline:))
		     ((= :vv: 7) (vl-cmdf "_layer" "on" "0,Defpoints" ""))
		     ((= :vv: 8) (vl-cmdf "_layer" "off" "Defpoints" ""))
		     ((= :vv: 9) (vl-cmdf "_layer" "lock" "Defpoints" ""))
		     ((= :vv: 10)(vl-cmdf "_layer" "unlock" "Defpoints" ""))
		     ((= :vv: 11)
		      (vl-cmdf "_layer" "on" "Defpoints" "unlock" "Defpoints" "thaw" "Defpoints" "")
		     ;(setq :ss: (ssget "X" '((0 . "XLINE") (8 . "Defpoints") (62 . 244))))
		      (setq :ss: (ssget	"X" '((-4 . "<or") (0 . "XLINE") (70 . 1) (0 . "RAY") (-4 . "or>") (8 . "Defpoints") (62 . 244))))
		      (if :ss: (vl-cmdf "_.erase" :ss: ""))
		     )
		     ((= :vv: 12) (:idgn_offset:))
	       )

               (princ)
  )

;;; What I want to say is that it seems to have something to do with WINDOWS Update
;;; But i'm not sure
;;; Because this situation started happening recently
;;; At present, all my CADs have been updated.

KuoTze, 2023-02-14:
> Can you provide me a simple script that allows me to repeat what you are seeing?

This problem
Mainly because the resolution of the extended monitor (4K) is different from that of the main monitor (2K)
Need to specify the work screen of AUTOCAD
avoid cross-monitor