dalefugier/DOSLib #22: dos_getfiled: Weird entries in text-filter

Pierre-de-la-Verre, 2022-01-28:
I realized it today, on Win 10, with Brics21 and Acad18: Sometimes the filter-list in dos_getfiled (where you filter *.dwg; *.dxf; *.txt or nothing) is amend with (asian?) characters or a mixture of it. I tried and got up to 4 lines - one correct, 3 wrong. Mostly it disappears the next time I use the command. Have you seen this behaviour before?
[image]

Pierre-de-la-Verre, 2022-04-12:
It seem to be a "timing problem"(??) when loading the program with autoload. The first time - when autoload is invoked and the program starts - it gets the strange data. Here an example when it displays the last "princ" which is also called with the autoload.
[image]

dalefugier, 2022-08-29:
Please provide code that doesn't work - thanks.

Pierre-de-la-Verre, 2022-09-19:
Hi Dale
sorry for my delay. Here is the code and the current screenshot. 
AutocadMap 2018, German
```
; ; ; ; Programm:   example.lsp
; ; ; ; Zweck:      Shows that 'dos_getfiled' gives a strange file selection when this lisp is called the first time in an AutoCAD Session.
; ; ; ; 
; ; ; ;  *********************************************;; 

(princ (strcat "\npunktsetzmode: " (vl-princ-to-string punktsetzmode)))
(princ (strcat "\nauswahlmode: " (vl-princ-to-string auswahlmode)))

(defun c:example ()
    (setq stopflag 0
          koord_liste nil
          koord_zaehler -1
          quelldatei (dos_getfiled "Koordinatenliste wählen, Trennzeichen beachten" (dos_tempdir) "All files (*.*)|*.*")
    )
)

```
[image]

dalefugier, 2022-09-19:
Your filter string should look like this:
```
"All files (*.*)|*.*||"
```
See the help file for details.

Pierre-de-la-Verre, 2022-09-19:
Thanks, Dale.

Should have seen it - sorry for the wrong "issue".
Peter

closed 2022-09-19 by Pierre-de-la-Verre