dalefugier/DOSLib #26: AutoCAD crash when using DOS_GETFILED or DOS_GETDATE

ConstantVariable6105, 2022-10-07:
I have a Windows 10 user, version 21H2, that has frequent AutoCAD crashes when using AutoLISP functions that contain the DOS_GETFILED and the DOS_GETDATE commands.

However, it does not crash every time. For example, reboot computer, use commands several times, without issues. User leaves AutoCAD running overnight. In the morning, first use of DOS_GETDATE results in AutoCAD crash. By crash I mean closed without warning or chance to save.

Today, user must have used DOS_GETFILED over a dozen times before it crashed AutoCAD.

No other user with similar systems are experiencing these crashes.

I am really struggling where to start debugging. I have considered uninstall / reinstall of AutoCAD.

I would welcome any advice.

Regards,
Jerry

dalefugier, 2022-10-11:
Hey @ConstantVariable6105, are you capable of providing a way to repeat any of this? Without a way, there probably isn’t much I can do.

ConstantVariable6105, 2022-10-13:
Hi, Dale.
At this time, no. I just wanted to see if anyone else was experiencing random crashes with DOSLIB and AutoCAD 2022.
I don't recall any of these problems when using the same tools in AutoCAD 2019.
If I am able to reproduce the crash, I will report.
Thanks,
Jerry

closed 2022-11-19 by dalefugier