dalefugier/DOSLib #31: dos_htmlbox not accepting certain characters in the url

irefugee, 2023-05-04:
I get several errors in a dialogue box when the url contains specific characters. Please try the following snippet inside autocad.

(dos_htmlbox "Street View" "https://www.google.com/maps/@32.9884474,-117.0756695,3a,75y,173.54h,57.13t/data=!3m7!1e1!3m5!1sRFjtZIFqCLBSNUIsbzgaAw!2e0!5s20200201T000000!7i16384!8i8192" 900 700)

CEXT-Dan, 2023-05-05:
what errors you you see? I can't event get to maps.. I tried to access another site and got
'Support for Internet Explorer ended on June 15, 2022'

CEXT-Dan, 2023-05-05:
> I get several errors in a dialogue box when the url contains specific characters. Please try the following snippet inside autocad.
>
> (dos_htmlbox "Street View" [...]

if you are using autocad, maybe you can use (showHTMLModalWindow "https://www.google.com/maps/@32.9884474,-117.0756695,3a,75y,173.54h,57.13t/data=!3m7!1e1!3m5!1sRFjtZIFqCLBSNUIsbzgaAw!2e0!5s20200201T000000!7i16384!8i8192")

irefugee, 2023-05-05:
> if you are using autocad, maybe you can use (showHTMLModalWindow [...]

Thanks! that works. Though the 
```
(setq url "https://www.google.com/maps/@?api=1&map_action=pano&viewpoint=32.92278146,-117.05944686&heading=0&pitch=0&fov=100\"")
(showhtmlmodalwindow url)
```
The errors have something to do with Object not supporting sc_initiPage,sc_initQueryParameterPreserver, scinitStats,
When I use a regular url without parameters it will load fine.
I do know how to attach screenshots but there are more than 10 script errors I just click Yes to continue to clear it.