dalefugier/DOSLib #32: dos_progbar doesn't count "relative"

Pierre-de-la-Verre, 2023-10-18:
I took the examples from the helpfile and made this small loop.
If I understand correctly it should set the value for prog_bar to 10 and then counts back to 0.
But I see the full prog_bar, see counting the loops, and finally the progbar is erased.

Using absolute value works fine.
What's wrong here?

`(defun c:checkdosprogbar ( )
    (setq counter 0)
    (dos_progbar "Wait..." 10)
    (dos_pause 3)
    (repeat 10
        (princ (strcat "\nCounter: " (itoa (setq counter (1+ counter)))))
        (dos_pause 3)
        (dos_progbar -1)
    )
    (dos_progbar)
    (princ)
)`

dalefugier, 2023-10-23:
Call this with a positive value within the range specified to set the current position of the status bar. Pass a negative number to add an amount to the current position (relative).

closed 2023-10-23 by dalefugier

Pierre-de-la-Verre, 2023-10-25:
Sorry, but I don't understand. What will be the right code from above to count from 0 to 10?

dalefugier, 2023-10-27:
This seems to work:
```
(defun c:dosbar ( )
    (setq counter 0)
    (dos_progbar "Wait..." 10)
    (dos_pause 3)
    (repeat 10
        (princ (strcat "\nCounter: " (itoa (setq counter (1+ counter)))))
        (dos_pause 1)
	(dos_progbar counter)
    )
    (dos_progbar)
    (princ)
)
```

Pierre-de-la-Verre, 2023-10-27:
Yes, that's the "absolute" version, when I understand it correctly.
Thanks.